kaluma-project/kaluma #92: Need to check all I/O handlers are freed when `.clear` command executed

niklauslee, 2019-04-07:
An example:

1. Run some `setInterval()`, it alloc `io_timer_handle_t`
2. Run `.clear` command
3. Need to check the allocated `io_timer_handler_t` is freed.

Need to check for all `io_***_handle_t` type are freed.

niklauslee, 2019-04-09:
This commit seems that I/O handles are not deallocated properly. `reset_io()`function just initialize I/O handle lists, but still remains I/O handle object in heap memory (need to use `free()` for all I/O handle object). As you know, all C objects allocated in heap memory by `malloc()` should be deallocated by `free()`.

To fix this problem properly, you need to understand exactly how I/O handles are created and managed in `io.h`, `io.c` as well as list structures and functions in `util.h`, `util.c`. (This idea is from [libuv](https://github.com/libuv/libuv) which used by Node.js to support asynchronous I/O). I can explain in detail at some time or fix by myself if you have not enough time to investigate. :blush:  Node.js architecture is quite complicated, so not easy to understand. :confounded:

P.S. To keep naming convention, `io_reset()` is preferred instead of `reset_io()`. (personally `io_clear()` is preferred)

communix, 2019-04-09:
> you have not enough time to investigate. 😊 Node.js architecture is quite

Good point.
I'll change the function name io_clear() and try to fix it again.

Thanks.
Jay Kim.

niklauslee, 2019-04-10:
`.clear` command name is changed to `.reset` for clarification

commit faaecf0f (2019-04-10): Fix #92, #93
closed 2019-04-10 by niklauslee