kaluma-project/kaluma #147: `.reset` stucks when error in flash user code.

niklauslee, 2019-08-17:
Steps to reproduce

1. Write below code in flash
```js
function f1() {
  f2(); // error
}
f1();
```
2. Reset board
3. Enter `.reset` command
4. Stuck

communix, 2019-08-18:
Root cause : hard fault at the jmem_pools_collect_empty function.
 
[image]

This code is called at the jerry_gc() in the runtime_init() function.
Let me take a look more.

niklauslee, 2019-08-22:
Problem not occur if `jerryxx_print_error(...)` is not called.

niklauslee, 2019-08-22:
```c
void jerryxx_print_error (jerry_value_t value, bool print_stacktrace) {
  jerry_value_t error_value = jerry_get_value_from_error (value, true); // <-- true parameter cause problem
  //  ...
  jerry_release_value (error_value);
}
```

The second parameter `jerry_get_value_from_error()` should be `false` if you will release the value `error_value` manually by calling `jerry_release_value()`

closed 2019-08-22 by niklauslee