kaluma-project/kaluma #165: System stuck when `.load` twice the example codes: https://kameleon.io/@niklauslee/ssd1306

niklauslee, 2019-09-19:


niklauslee, 2019-09-28:
This seems related with [#3172](https://github.com/jerryscript-project/jerryscript/issues/3172) (I'm not sure)

niklauslee, 2019-10-02:
Confirmed that `jerry_cleanup()` will call free callback functions. Need to check there is no reference from C API to the object having `gc_handle`.

niklauslee, 2019-10-02:
More simplified bug reproduce code (Try in playground):

```js
var BufferedGC = require('graphics').BufferedGraphicsContext;
var gc = new BufferedGC(128, 64, {
  display: (buffer) => {}
});
global.gc = gc;
```

The above code do not call free callback when `.reset`. But if removing display function, free callback successfully called when `.reset`

```js
var BufferedGC = require('graphics').BufferedGraphicsContext;
var gc = new BufferedGC(128, 64, {
  // display: (buffer) => {}
});
global.gc = gc;
```

niklauslee, 2019-10-02:
This issue is resolved. There were many reasons causing this problems:

1. Try to release value of function argument especially in the places using `JERRYXX_GET_ARG_OPT()`. It shouldn't. Now `JERRYXX_GET_ARG_OPT()` has removed, so `JERRYXX_GET_ARG()` should be used with checking `JERRYXX_HAS_ARG()`. This problem were found in many native modules I2C, SPI, etc. __Actually this problem caused system stuck.__
2. The callback functions in `gc_handle` struct (`display_js_cb`, etc.) prevented GC. So now move them into the class member variables (e.g. `this.__display_cb`) so as to allow GC.
3. `setTimeout()` didn't release values even when it timed out. This also prevented GC.

__Exact matching `jerry_get/create_*` and `jerry_release_value` is very important__

closed 2019-10-02 by niklauslee

communix, 2019-10-03:
Great job!! Thank you for your hard work!

commit 98d96163 (2019-10-02): Fix #165