kaluma-project/kaluma #172: Heap overflow before garbage collection is performed

niklauslee, 2019-10-01:


niklauslee, 2019-10-14:
Seems the same problem with the item no.2 described in https://github.com/kameleon-project/kameleon/issues/165#issuecomment-537433707

niklauslee, 2019-10-14:
A simple possible solution is just calling `jerry_gc()` after the timer js callback function. But it may cause a bit  (or severe?) performance degradation.

communix, 2019-10-14:
I tried to call jerry_gc() every 10 sec. but the lock-up issue is not fixed. I think there's a memory leakage in the code.

niklauslee, 2019-10-14:
OK. It seems that referencing to js value from C native code by using `jerry_acquire_value()` causing the problem. It was the same problem in the #165. I will try to find a way to avoid referencing timeout callback function in C native code.

niklauslee, 2019-10-14:
There are more same things :(

- `timer_js_cb`
- `watch_js_cb`
- `read_js_cb`

niklauslee, 2019-10-15:
https://github.com/jerryscript-project/jerryscript/issues/3218

niklauslee, 2019-10-15:
Avoiding reference to callback in C code side using `jerry_acquire_value()` do not resolve this issue. 😭

Below code is works well without heap usage increasing:

```js
setInterval(function () {
  console.log ("Current time is unknown");
}, 100);
```

closed 2019-10-15 by niklauslee

HotCodeworker, 2019-10-16:
I've known that this problem is originally from mcu's limitted performance. Am I wrong? If I set 10ms to time when using timer API, I wonder how the problem occures. please understand me!

niklauslee, 2019-10-16:
@HotCodeworker This problem is about heap memory management, not performance. There was memory leak preventing GC. If you set very small time interval, the callback function will not be executed for every time interval instead of system stuck.

commit b73ef500 (2019-10-15): Fix: #172