kaluma-project/kaluma #181: Add `dgram` module

niklauslee, 2019-10-15:


tunikakeks, 2022-09-07:
Please add I really need it

Apollon77, 2023-08-27:
Adding this would allow (maybe with some more features) to use it to build matter compatible devices …

glennswest, 2023-10-26:
Just wonderfing what's the plan. I dont mind working on it, just dont want to interrupt a development stream if its in progress.
It's of course two parts, one is dgram module, and second is enhancements that are needed in cyw43 if any. And a additional would be examples. I can certianly fork, work it out, and then do a merge. Let me know.

glennswest, 2023-10-26:
Reviewing the wifi driver side, it looks good to go. Seems just missing dgram module at a glance.
Wondering on max socket limititations.

niklauslee, 2023-10-30:
@communix

communix, 2023-10-30:
@glennswest I didn't start to implement dgram module so please implement it. code contribution is always welcome.
Upd device driver of cyw43 is implemented but not fully tested. (https://kalumajs.org/docs/api/device-driver#network-device-driver)