kaluma-project/kaluma #370: Need to check the frequency `tone()`

niklauslee, 2021-03-24:
When plays some musical notes, it sounds somewhat weird.
Example: https://kaluma.io/@kaluma/buzzer

communix, 2021-03-25:
I checked all the frequency without buzzer connection. The PWM frequency is right.

C4, Frequency 262Hz
[image]

D4 Frequency 294Hz
[image]

E4 Frequency 330Hz
[image]

F4 Frequency 370Hz
[image]

G4 Frequency 392Hz
[image]

A4 Frequency 440Hz
[image]

B4 Frequency 466Hz
[image]

C5 Frequency 523Hz
[image]

communix, 2021-03-25:
I think the issue is the buzzer, If the buzzer draw a high current, the pico PWM can't draw the enough current. So the PWM signal is distorted like the below signal and it may cause the issue. So please try it with the active buzzer.

[image]

communix, 2021-03-25:
I tried it another buzzer and it make more noisy sound and here's the PWM signal

[image]

niklauslee, 2021-03-25:
Works well with buzzer for 3.3V, so buzzers for 3.3V should be used to play a correct melody.

closed 2021-03-25 by niklauslee