kaluma-project/kaluma #390: Support Bluetooth API

niklauslee, 2021-04-16:
Ref: [Web Bluetooth API](https://developer.mozilla.org/en-US/docs/Web/API/Web_Bluetooth_API)

tannewt, 2021-08-03:
FYI we've defined a file transfer BLE service: https://github.com/adafruit/Adafruit_CircuitPython_BLE_File_Transfer We're using it for CircuitPython but the API is generic. I've started a webbluetooth library for it here: https://github.com/adafruit/ble-file-transfer-js

niklauslee, 2021-08-04:
@tannewt Thanks for your info!

dashersw, 2023-08-06:
Hi team! I recently found out about kaluma and I’m deeply in love with it! I have a use case that I want to build for BLE, and I just found out that it’s not supported yet. I looked into the stack to see whether I could write the Bluetooth module myself but it seems a little complicated for the uninitiated. Do you have any plans for Bluetooth support or some starter code for me to pick up and build upon?

montanag, 2023-09-12:
Also curious on this. I just coded up a IOT device using micropython but would like to use kaluma instead. This is a show stopper for many of my applications. Is there any way I can contribute to help implement this?

Mattijah, 2025-01-20:
Same here... can't believe it hasn't been resolved in almost 4 years