kaluma-project/kaluma #412: Support API for power saving

niklauslee, 2021-07-28:


closed 2021-08-29 by niklauslee

ishiharas, 2022-07-01:
hello,  sorry for getting back on this old ticket, but I have a short question. In the docs I only see the dormant() function for triggering sleep with an external source. Is there a way to let the device sleep with just the sleep timer dependent on the internal clock, like described here?

https://github.com/raspberrypi/pico-extras/blob/master/src/rp2_common/pico_sleep/sleep.c

communix, 2022-07-02:
@ishiharas you can use delay() function.

[image]

This delay function is using sleep_ms() API for RP2 and this function make a system to go sleep (WFE) mode.

[image]