kaluma-project/kaluma #413: Enhance tone() API

niklauslee, 2021-08-01:
- Move `duration` parameter to the next of `duty`
- Function for pull/push with two GPIO pins (e.g. [`toneAC()`](https://github.com/teckel12/arduino-toneac) in Arduino)

communix, 2021-08-05:
@niklauslee I upload the code with inversion function. please verify the image in the /develop/enhance_tone branch.
I added invPin in the tone API.

tone(pin[, frequency[, duration[, duty [, invPin]]])

For the RP2040, invPin must be the same PWM slot but different channel.

[image]

tone(0, 300, 1000, 0.5, 1); // Generate 300Hz tone on the pin 0 and pin 1 (inversion) for 1sec.
pin 0 and pin 1 are the same slot (0) and pin0 is channel A (0A) and pin1 is channel (0B)
Pin17 can be used for the invPin of pin0 because Pin17 is slot 0, channel B.

Please verify the function and let me know if you find any bug in this feature.

communix, 2021-08-05:
This is the signal when I run the below code.
tone(0, 300, 1000, 0.5, 1); // Generate 300Hz tone on the pin 0 and pin 1 (inversion) for 1sec.

Channel 1(Yellow): Pin0
Channel 3(Blue): Pin1

[image]

niklauslee, 2021-08-05:
It works! Very nice!

niklauslee, 2021-08-09:
Optional parameters are too long, so here I suggest the signature of `tone()` function as below:

### tone(pin[, frequency[, options]])
- `pin`
- `frequency`
- `options`
  - `duration`
  - `duty`
  - `invert` `<number>` __Default: `-1`__. GPIO pin number where to generate inverted signal. `-1` means no inverted signal. ...more description...

closed 2021-08-15 by niklauslee