kaluma-project/kaluma #496: REPL improvements

davidmoshal, 2022-02-21:
1) ability to enter multi-line statements.

[image]

2) tab completing (like EspruinoJS).
-  wondering if it's possible to get this info from Jerryscript.

niklauslee, 2022-02-22:
About 2, is that works only for predefined funs and vars? Or, works exactly like typical code completion in VSCode, etc.?

davidmoshal, 2022-02-24:
Works exactly like VSCode.
Any new var / const can be tab-completed.
You should try it out.
If we had that working, then I'd be willing to work on a user interface for Kaluma similar to Espruino.

niklauslee, 2022-02-25:
It doesn't seem easy with JerryScript API. Any ideas are welcome.

davidmoshal, 2022-02-25:
I'm no C expert sadly, maybe the answer is in the Espruino codebase.

Looking at the Jerryscript api, there is the ability to look at types and enumerate their properties.
https://jerryscript.net/api-reference/#jerry_object_type_t

There is also a context:
https://jerryscript.net/api-reference/#jerry_context_t

So. presumably given the context, and the ability to enumerate the properties of an object tab completion should be possible, no?

niklauslee, 2022-02-26:
For the feature I think it is required to build partial AST (parsing incomplete JS expression).