kaluma-project/kaluma #501: Problems with rtc

Runc2333, 2022-02-28:
(new Date()).getMilliseconds() Always returns 0, (new Date()).getTime() also won't return the correct unix timestamp, the last three digits is always 0.
Passing a unix timestamp in format of milliseconds to rtc.setTime() just simply doesn't work, but passing a timestamp in format of seconds will working, this behaves opposite to the docs.

niklauslee, 2022-02-28:
That's because Pico's RTC API works only in second-level. For milliseconds, please try `millis()`.

niklauslee, 2022-03-02:
@Runc2333  It seems works well. Please explain in detail how it doesn't work.

```
> .hi
         ___
   _____/ . \   Welcome to Kaluma v1.0.0-beta.13
  /        __>  https://kaluma.io
 |/\________/
  \__@  v  v    Type '.help' for more info.
-----------------------------------------------

> const rtc = require('rtc')
undefined
> rtc.getTime()
404000
> rtc.getTime()
406000
> rtc.getTime()
406000
> const c = 1000 * 60 * 60 * 24 * 30; // 30 days
undefined
> c
2592000000
> rtc.setTime(c)
undefined
> rtc.getTime()
2592004000
> rtc.getTime()
2592005000
> rtc.getTime()
2592006000
> rtc.getTime()
2592006000
> new Date()
{}
> (new Date()).toString()
'Sat Jan 31 1970 00:00:19 GMT+0000'
> (new Date()).toString()
'Sat Jan 31 1970 00:00:21 GMT+0000'
> (new Date()).toString()
'Sat Jan 31 1970 00:00:22 GMT+0000'
> (new Date()).toString()
'Sat Jan 31 1970 00:00:23 GMT+0000'
> (new Date()).toString()
'Sat Jan 31 1970 00:00:24 GMT+0000'
> (new Date()).toString()
'Sat Jan 31 1970 00:00:29 GMT+0000'
> (new Date()).toString()
'Sat Jan 31 1970 00:00:30 GMT+0000'
> (new Date()).toString()
'Sat Jan 31 1970 00:00:36 GMT+0000'
> (new Date()).toString()
'Sat Jan 31 1970 00:00:38 GMT+0000'
> rtc.setTime(0);
undefined
> (new Date()).toString()
'Thu Jan 01 1970 00:00:02 GMT+0000'
> (new Date()).toString()
'Thu Jan 01 1970 00:00:03 GMT+0000'
> (new Date()).toString()
'Thu Jan 01 1970 00:00:04 GMT+0000'
> (new Date()).toString()
'Thu Jan 01 1970 00:00:05 GMT+0000'
> 
```

Runc2333, 2022-03-02:
Please try this out.
`> .hi
         ___
   _____/ . \   Welcome to Kaluma v1.0.0-beta.13
  /        __>  https://kaluma.io
 |/\________/
  \__@  v  v    Type '.help' for more info.
-----------------------------------------------

> const rtc = require('rtc');
undefined
> rtc.getTime();
17000
> rtc.setTime(1646219452553) // Wed Mar 02 2022 19:10:52 GMT+0800
undefined
> rtc.getTime()
70000
> rtc.setTime(parseInt(1646219452553 / 1000))
undefined
> rtc.getTime()
1646219454000`

Runc2333, 2022-03-02:
Please try this out.
```
> .hi
         ___
   _____/ . \   Welcome to Kaluma v1.0.0-beta.13
  /        __>  https://kaluma.io
 |/\________/
  \__@  v  v    Type '.help' for more info.
-----------------------------------------------

> const rtc = require('rtc');
undefined
> rtc.getTime();
17000
> rtc.setTime(1646219452553) // Wed Mar 02 2022 19:10:52 GMT+0800
undefined
> rtc.getTime()
70000
> rtc.setTime(parseInt(1646219452553 / 1000))
undefined
> rtc.getTime()
1646219454000
```

niklauslee, 2022-03-02:
It's so strange! It works well on my board.

```
> .hi
         ___
   _____/ . \   Welcome to Kaluma v1.0.0-beta.13
  /        __>  https://kaluma.io
 |/\________/
  \__@  v  v    Type '.help' for more info.
-----------------------------------------------

> const rtc = require('rtc');
undefined
> rtc.getTime();
15000
> rtc.setTime(1646219452553)
undefined
> rtc.getTime()
1646219458000
> (new Date()).toString()
'Wed Mar 02 2022 11:11:09 GMT+0000'
>
```

niklauslee, 2022-03-02:
@Runc2333 Can you try on another board?

Runc2333, 2022-03-02:
@niklauslee Sorry but this is the only board I have, I'll purchase a new one and try.
Keep in touch.

niklauslee, 2022-03-02:
@Runc2333 I've just tried on another board. I have the same problem. I'll investigate this problem.

Runc2333, 2022-03-02:
@niklauslee Great, looking forward to the result.

niklauslee, 2022-03-03:
@Runc2333 Same issue with #488
I forgot the issue was fixed after 1.0.0-beta.13 release, so it will be included in the next release.
Current `master` branch build works well.

closed 2022-03-03 by niklauslee

niklauslee, 2022-03-07:
@Runc2333 Fixed in `1.0.0-beta.14` and now released.

Runc2333, 2022-03-07:
@niklauslee Thank you for your notice, I'll try that.