kaluma-project/kaluma #504: Cannot access VSYS(GPIO29, ADC3) on rp2

Runc2333, 2022-03-07:
According to Raspberry Pi Pico's official documentation...
```
A few RP2040 GPIO pins are used for internal board functions, these are:
GPIO29 IP Used in ADC mode (ADC3) to measure VSYS/3
```
But I can't use `analogRead(29)` or `board.adc(29).read()` to get the voltage at VSYS.
Could you help me with that?

Runc2333, 2022-03-07:
Also, I mentioned that if I flash the rp2 a few times (maybe 5~6 times) without the power cut, the board will hang up until you reconnect it's power. Is this a bug or is it just designed this way?

niklauslee, 2022-03-07:
@communix Can you comment on this?

Runc2333, 2022-03-08:
@niklauslee I build a version of kaluma which changes `targets/rp2/src/adc.c` from
```
if ((pin >= 26) && (pin <= 28)) {
  return pin - 26;  // GPIO 26 is channel 0
}
```
to
```
if ((pin >= 26) && (pin <= 29)) {
  return pin - 26;  // GPIO 26 is channel 0
}
```
and it just works.
`board.adc(29).read() * 10 - 0.1` equals the voltage at VSYS.

Runc2333, 2022-03-08:
I created a pull request for this, you can review it to check if it works.

closed 2022-03-14 by niklauslee