kaluma-project/kaluma #513: any plan of support rasberry-pi-pico-w board ?

jjpark78, 2022-07-02:
rasberry pi  just release their new board that support wifi. 

https://www.raspberrypi.com/news/raspberry-pi-pico-w-your-6-iot-platform/

do we have any plan of support this new board ??

niklauslee, 2022-07-05:
Sorry I have no time to do it at this time. I want to encourage voluntary contributions. 😅

dwyschka, 2022-07-06:
I've got 3 of them. 
The Basic Image is working, but we need some other pin definitions. Because the onboard-LED is on the wifi-chip, not on the pico board anymore.

I havent testet wifi yet. (but i got problems to compile kaluma).
I'm a beginner with it, but would try my best

sergeychernov, 2022-07-08:
I have a problem to compile/build kaluma too.
I have tried this way:
`node ./build.js --target rp2 --board pico`
And received:
`make: *** No targets specified and no makefile found.  Stop.`

communix, 2022-07-17:
@sergeychernov please find the below link and try it.
https://github.com/kaluma-project/kaluma/wiki/Build

adrian-burlacu-software, 2022-09-11:
> @sergeychernov please find the below link and try it. https://github.com/kaluma-project/kaluma/wiki/Build

This works great!! I flashed it to my pico w and I can see the `process.binding('pico_cyw43').PICO_CYW43` function.

Can you please post an API example? All I can see from the change is `PICO_CYW43.prototype.close()`.

communix, 2022-10-15:
@adrian-burlacu-software Thank you for your feedback.

The first issue is known issue #531.
Let me check the other issues and fix it soon. 

And let's discuss this topic in this link (https://github.com/kaluma-project/kaluma/discussions/515) 

I'll close this issue since we already have a plan to support pico-W.

closed 2022-10-15 by communix

communix, 2022-10-15:
@adrian-burlacu-software 
Please create issue with the [PICO-W] prefix when you find any issue.
Thank you so much for your help.