kaluma-project/kaluma #542: Raspberry Pi Pico W support?

TSel1209, 2022-12-03:
Hello. I was trying to use the WiFi module provided by Kaluma with the Pico W's built-in wireless interface. It threw an error and said that an interface was not detected. Could support possibly be added for this?

thefreymaster, 2022-12-13:
The author is already working on it.  See this branch -> https://github.com/kaluma-project/kaluma/tree/pico-w

c6fc, 2023-01-03:
It looks like the pico-w branch was merged a few days ago, but using the newly-built pico-w .uf2, I still get the same error as on the pico build.

[image]

Would it be possible to get updated documentation on how to use the pico-w board and CYW43 driver with Kaluma? And potentially a pico-w release?

closed 2023-01-04 by niklauslee

niklauslee, 2023-01-04:
@c6fc Please try with the 1.1.0-beta.1 version.

shreeve, 2023-05-11:
@niklauslee - Is there a quick way to see what modules have been included in a particular build? For example, if I run `kaluma shell`, can I type something that will show me which modules are defined and available? If I create a new `*.uf2` file, how can I make sure that I include the right modules when I build the file? What are the command line arguments to ensure that I have, say, the cyw43 module?