kaluma-project/kaluma #546: Support for disconnected SPI pins

dseymore, 2023-01-01:
I was working with the [Waveshare RP2040-LCD1.28 board](https://www.waveshare.com/wiki/RP2040-LCD-1.28) and had to make a tweak in Kaluma to not define the `mosi` pin, as the default of `pin 12` on this board was actually the reset for the display. I needed to define SPI1 as:

```js
new SPI(1, {
  mode: SPI.MODE_0,
  sck: 10,
  mosi: 11,
  miso: -1,  //expected to be 8 or 12, but not connected on this board
  baudrate: 40_000_000,
  bitorder: SPI.MSB
});
```

The diff that worked around this for me, but isn't really ready for a pull request:

```diff
diff --git a/targets/rp2/src/spi.c b/targets/rp2/src/spi.c
index d91f79a..110c23b 100644
--- a/targets/rp2/src/spi.c
+++ b/targets/rp2/src/spi.c
@@ -31,7 +31,8 @@ struct __spi_status_s {
 } __spi_status[SPI_NUM];
 
 static bool __check_spi_pins(uint8_t bus, km_spi_pins_t pins) {
-  if ((pins.sck < 0) || (pins.miso < 0) || (pins.mosi < 0)) {
+  if ((pins.sck < 0) ||
+      (pins.mosi < 0)) {
     return false;
   }
   if (bus == 0) {
@@ -45,7 +46,7 @@ static bool __check_spi_pins(uint8_t bus, km_spi_pins_t pins) {
       return false;
     }
   } else if (bus == 1) {
-    if ((pins.miso != 8) && (pins.miso != 12)) {
+    if ((pins.miso != -1) && (pins.miso != 8) && (pins.miso != 12)) {
       return false;
     }
     if ((pins.mosi != 11) && (pins.mosi != 15)) {
@@ -147,7 +148,9 @@ int km_spi_setup(uint8_t bus, km_spi_mode_t mode, uint32_t baudrate,
   }
   spi_init(spi, baudrate);
   spi_set_format(spi, 8, pol, pha, order);
-  gpio_set_function(pins.miso, GPIO_FUNC_SPI);
+  if (pins.miso != -1) {
+    gpio_set_function(pins.miso, GPIO_FUNC_SPI);
+  }
   gpio_set_function(pins.mosi, GPIO_FUNC_SPI);
   gpio_set_function(pins.sck, GPIO_FUNC_SPI);
   if (miso_pullup) {
```

niklauslee, 2023-01-02:
@communix Any comments about this issue?

communix, 2023-01-03:
@dseymore Here's the new code to fix this issue. (https://github.com/kaluma-project/kaluma/tree/develop/spi_pin_issue)
Could you please confirm this code fixes your code?

dseymore, 2023-01-04:
Worked for me! Thank you

closed 2023-01-04 by niklauslee