kaluma-project/kaluma #549: Incorrect socket details for 'net' and 'http' on pico-w board

c6fc, 2023-01-04:
Using the [kaluma-rp2-pico-w-1.1.0-beta.1.uf2](https://github.com/kaluma-project/kaluma/releases/download/1.1.0-beta.1/kaluma-rp2-pico-w-1.1.0-beta.1.uf2) release, the 'createServer' method for 'net' and 'http' modules shows incorrect details for connected sockets:

[image]

In both cases, calls to '.write()' methods fail to deliver the data back to the client, but no errors are returned and the callbacks for both '.write()' and '.end()' execute normally. I haven't been able to verify what the outbound packet from the pico-w look like (so I don't know if injecting the correct raddr and rport will cause the data to be delivered), but at the moment it seems that the incorrect socket details are preventing server responses from reaching the client.

communix, 2023-01-09:
@c6fc Could you share the source code you tested? I want to reproduce the same issue on my setup.

c6fc, 2023-01-09:
Sure thing. These are both after a lot of iterations during testing, so they're super ugly, but it should be reproducible for you:

https://gist.github.com/c6fc/49d27fb60b8313029d756e497a6b7bfd
https://gist.github.com/c6fc/dd95af62477d184a96863d496603e469

communix, 2023-01-19:
@c6fc It's strange. there's no changes related to the wifi driver. 
Could you please try it with .reset command before .load command?

And please use attached uf2 file which I used for this test.

[kaluma-rp2-pico-w-1.1.0-beta.1_pico_w_test_0118.uf2.zip](https://github.com/kaluma-project/kaluma/files/10453325/kaluma-rp2-pico-w-1.1.0-beta.1_pico_w_test_0118.uf2.zip)

etx, 2023-02-06:
@communix I just gave this latest uf2 a try. `net.createServer` seems to work correctly once. I can do a res.write to send a string back to the client (netcat in this case). After that one connection I have to hard reset to make another connection.  Unfortunately `http.createServer` is hanging when I call it. 

For both I'm just using the example code from the docs.

Appreciate your work on this!!

emerson-a, 2023-02-12:
I have the same issue.  

Here's my code: https://gist.github.com/emerson-a/d48df41f8431ebcb3ed516f17485d7ef

I also appreciate your work on this!

communix, 2023-02-12:
@etx @emerson-a Thank you for your feedback. let me check it.

communix, 2023-02-13:
@emerson-a @etx 
Could you please try it with this binary? http.create server is working with this code.
But I can't reproduce the hard reset issue, (After that one connection I have to hard reset to make another connection.) If you still have the same issue, please create new issue with the detailed scenario.

[kaluma-rp2-pico-w-1.1.0-beta.1_pico_w_test_0212.uf2.zip](https://github.com/kaluma-project/kaluma/files/10719730/kaluma-rp2-pico-w-1.1.0-beta.1_pico_w_test_0212.uf2.zip)

K4ST0R, 2023-02-18:
I got the same issue than @emerson-a and @etx .

After one connection, I can't make another connection.

I try to debug it but I am really struggling. It seems to have a link with the code using the listen fd to manage the connection with the client.

Changing : 
https://github.com/kaluma-project/kaluma/blob/192154ee7f4dba8421b026c97448910c850bb438/src/modules/pico_cyw43/module_pico_cyw43.c#L680-L683
by
```c
static err_t __tcp_server_accept_cb(void *arg, struct tcp_pcb *newpcb,
                                    err_t err) {
  if (err == ERR_OK)  {
 // test create fd
  uint8_t *oldFd = (uint8_t *)arg;
  int fdV = km_get_socket_fd();
  uint8_t *fd = &fdV;
  memcpy(&__socket_info.socket[*fd], &__socket_info.socket[*oldFd], sizeof(__socket_data_t));
  __socket_info.socket[*fd].fd = fdV;
```
(Sorry for the really bad code!)

I am able to connect again with another client but i can't connect two client in parallel.

emerson-a, 2023-02-18:
> @emerson-a @etx Could you please try it with this binary? http.create server is working with this code. But I can't reproduce the hard reset issue, (After that one connection I have to hard reset to [...]

I have tested this but the same issue occurs.  I can only make one http connection.

Thanks again for your work on this!

communix, 2023-02-19:
@emeson-a, @K4ST0R Thank you for your feedback. I think the remaining issue issue is two client support. I have created new issue about is (#557), So let's discuss this issue at #557. I'll close this issue.
If anyone find new issue or the issue which is not fixed now, please create new issue. I'll take a look this issue.

closed 2023-02-19 by communix

etx, 2023-02-19:
@communix Could you post the JS file you're flashing to test http.createServer? I'm still having a hard time getting it to work. Thank you!

communix, 2023-02-19:
@etx I'm testing with the @emerson-a 's code. https://gist.github.com/emerson-a/d48df41f8431ebcb3ed516f17485d7ef