kaluma-project/kaluma #554: Ability to get IP address of Pico W?

TylerSelden, 2023-02-01:
Hello. I've started an IoT project using Kaluma and a Pico W. I quickly realized that there's no way to get the IP address of the Pi through Kaluma, which would be helpful as a new feature.

AcousticTypewriter, 2023-02-01:
@TylerSelden Hey there! This feature is already implemented. On both tcp and http server, you can get the ip address from server._dev.ip
This is definitely not well documented, but I based it off of an unrelated issue that showed the ip address.
Http: https://gist.github.com/c6fc/dd95af62477d184a96863d496603e469
Tcp: https://gist.github.com/c6fc/49d27fb60b8313029d756e497a6b7bfd

You could also download a network scanning app to find the pico's ip. I use Fing on my android and it identifies my pico very well.

Hope this helps!

communix, 2023-02-06:
I fixed some bug related to the network of Pico-W module.
Please use the latest code. (or you can download binary in this link<https://github.com/kaluma-project/kaluma/issues/549>

closed 2023-04-09 by communix