kaluma-project/kaluma #555: ADC4 Temperature

susnick, 2023-02-08:
I am trying to get temperature based on this article https://www.electroniclinic.com/read-raspberry-pi-pico-onboard-temperature-sensor/

In short:
Now, you might be thinking to which pin this onboard temperature sensor is connected? This onboard temperature sensor is connected to one of the ADCs or Analog-to-Digital Converters. The temperature sensor does not have a physical pin in the board but is accessed as ADC4. This onboard temperature sensor works by delivering a voltage to the ADC4 pin that is proportional to the temperature.

I am not sure how to access this based on available documentation. Any help is appreciated.

communix, 2023-02-13:
@susnick Hi, Kaluma does not support onboard temperature sensor now. I'll implement this feature in the next release. I think I can release test version soon.

communix, 2023-02-19:
@susnick Could you please try it with the attached test FW?
Here's the example code.
If you confirm it works well and close this issue, I'll merge this change into the main branch.

[kaluma-rp2-pico-1.1.0-beta.1_temp_adc_test_0218.uf2.zip](https://github.com/kaluma-project/kaluma/files/10776043/kaluma-rp2-pico-1.1.0-beta.1_temp_adc_test_0218.uf2.zip)
```
const { ADC } = require('adc');
const rp2 = require('rp2');

var a = new ADC(rp2.TEMPERATURE_ADC);
var value = a.read(); // Read the ADC value at the RP2 intenal temperature ADC.
var adc_voltage = value * 3.3;
var temperature = 27-(adc_voltage-0.706)/0.001721;
console.log(temperature);

```

susnick, 2023-02-20:
Looks good thanks.

closed 2023-02-20 by susnick

communix, 2023-02-25:
@susnick Thank you. I'll merge it to master branch and release it next release.