kaluma-project/kaluma #556: How to use SPI like in MicroPython ?

AhmedM-Dev, 2023-02-14:
I am trying to clone the MicroPython NRF24L01 (https://github.com/iottrends/NRF24l01_MicroPython/blob/main/nrf24l01.py) driver using KalumaJS.

My issue is that I could not replicate the exact behavior of **[spi.readinto](https://docs.micropython.org/en/latest/library/machine.SPI.html)** using KalumaJS SPI interface.

Can someone guide me how to do it ?

communix, 2023-02-17:
@AhmedM-Dev 
I checked the MicroPython spi.readinto function and I think it can be implemented by the spi.transfer() function of Kaluma.

[image]

Both function read and write the data through SPI.
Here's the difference.
 - MicroPython spi.readinto(buf, write=0x00) function read n byte (length of the buff size) data via SPI and store read data into buff. at the same time, it send "write"(byte) n times via SPI.
 - Kaluma spi.transfer(data [, timeout]) function write the n byte data (n is the length of the data) and return the read data.

In this example, the self.buf = bytearray(1), that means it's one byte buffer, so spi.readinfo send and receive only one byte.
Here's the code for Kaluma
 - MircoPython: readinto(self.buf, reg)
 - Kaluma: buf = spi.transfer(new Uint8Array([reg])); // reg is one byte data to send and buf is the read data(byte)

Thanks.

AhmedM-Dev, 2023-02-17:
@communix
Thank you so much for your well explained reply. I will try to implement the **nRF24L01** driver based on the MicroPython driver and send you feedback as soon as possible.

Thanks.

closed 2023-04-09 by communix