kaluma-project/kaluma #557: [pico-w] Multi client is not support in the http module

communix, 2023-02-19:
Only one client is working in Http module. Need to support multiple client.

communix, 2023-02-19:
It's from [K4STOR](https://github.com/K4ST0R)

I got the same issue than @emerson-a and @etx .

After one connection, I can't make another connection.

I try to debug it but I am really struggling. It seems to have a link with the code using the listen fd to manage the connection with the client.

Changing :
[kaluma/src/modules/pico_cyw43/module_pico_cyw43.c](https://github.com/kaluma-project/kaluma/blob/192154ee7f4dba8421b026c97448910c850bb438/src/modules/pico_cyw43/module_pico_cyw43.c#L680-L683)

Lines 680 to 683 in [192154e](https://github.com/kaluma-project/kaluma/commit/192154ee7f4dba8421b026c97448910c850bb438)

 static err_t __tcp_server_accept_cb(void *arg, struct tcp_pcb *newpcb, 
                                     err_t err) { 
   if (err == ERR_OK) { 
     uint8_t *fd = (uint8_t *)arg; 

by
static err_t __tcp_server_accept_cb(void *arg, struct tcp_pcb *newpcb,
                                    err_t err) {
  if (err == ERR_OK)  {
 // test create fd
  uint8_t *oldFd = (uint8_t *)arg;
  int fdV = km_get_socket_fd();
  uint8_t *fd = &fdV;
  memcpy(&__socket_info.socket[*fd], &__socket_info.socket[*oldFd], sizeof(__socket_data_t));
  __socket_info.socket[*fd].fd = fdV;
(Sorry for the really bad code!)

I am able to connect again with another client but i can't connect two client in parallel.

communix, 2023-02-26:
I realized that current code dose not support multi client and change the code.

communix, 2023-02-27:
[@emerson-a](https://github.com/emerson-a), [@etx](https://github.com/etx)

I just push new code to support multi-client, could you please test the issue with this binary?
[kaluma-rp2-pico-w-1.1.0-beta.1_test_0226.uf2.zip](https://github.com/kaluma-project/kaluma/files/10835099/kaluma-rp2-pico-w-1.1.0-beta.1_test_0226.uf2.zip)

But the system is unstable so it's hang up after 3~4 tries. I don't know why this happens now. Could you please check this and share your test result?

etx, 2023-02-27:
> [@emerson-a](https://github.com/emerson-a), [@etx](https://github.com/etx)
>
> I just push new code to support multi-client, could you please test the issue with this binary? [...]

Neat! It does perform the same way for me that you are describing. I can start up an HTTP server and make requests from two different clients on my LAN. After ~5 requests it does hang the Pico. But a step in the right direction. Happy to see you're working on this, thank you!

communix, 2023-03-12:
I'm still working on this issue and I hope I can find a solution soon.

communix, 2023-03-27:
I change the lwip settings and http server seems to be stable with it. Please try it.
I still see system hang with this change one time. So I don't think it's fixed but way better than before.
I'll test this FW and try to find a solution of new system hang issue.

[kaluma-rp2-pico-w-1.1.0-beta.1_test_0326.uf2.zip](https://github.com/kaluma-project/kaluma/files/11073959/kaluma-rp2-pico-w-1.1.0-beta.1_test_0326.uf2.zip)

communix, 2023-04-10:
@etx @emerson-a I think I fixed the system hang issue. Could you please try it with the latest pico-w FW?

[kaluma-rp2-pico-w-1.1.0-beta.1_http_test_0409.uf2.zip](https://github.com/kaluma-project/kaluma/files/11188052/kaluma-rp2-pico-w-1.1.0-beta.1_http_test_0409.uf2.zip)

etx, 2023-04-11:
> @etx @emerson-a I think I fixed the system hang issue. Could you please try it with the latest pico-w FW?
> 
> [kaluma-rp2-pico-w-1.1.0-beta.1_http_test_0409.uf2.zip](https://github.com/kaluma-project/kaluma/files/11188052/kaluma-rp2-pico-w-1.1.0-beta.1_http_test_0409.uf2.zip)

Thanks for pinging me on this! It does seem to be working much better now. I'm keeping a browser tab open and refreshing and it will work the majority of the time. Sometimes the browser does not get a response back and I see this in the terminal:
```
[CYW43] got unexpected packet -9
[CYW43] got unexpected packet -9
```

Maybe it has something to do with my wifi network. I'm using a google nest mesh system. I'll try another wifi setup to see if that is any more stable.

communix, 2023-04-15:
@etx I have the same issue and I'm using Google nest mesh system too. I didn't see the unexpected packet -9 error message but CYW lib call back function is called with error. So I think it's a network issue. There's nothing I can do now. Please let me know if you find anything about this issue.

communix, 2023-04-30:
@etx Thank you for your confirmation. Let me close this issue.

closed 2023-04-30 by communix

shreeve, 2023-05-12:
@communix - Hello! Thanks for you work on this... given that there are issues with very basic network communication, I suppose that MQTT support is not an option yet, is that right?