kaluma-project/kaluma #559: rp2 module does not work on pico-w

ZIMkaRU, 2023-02-26:
Pico-w just hangs when flashing, if only to add to code the export of the module, for example:
```js
const { ASM } = require('rp2')
```

Any thoughts?

communix, 2023-02-27:
@ZIMkaRU Let me check this issue.

communix, 2023-02-27:
@ZIMkaRU I checked this issue. pico board has no issue but pico-w has the issue. It's strange. I'll debug this issue.

communix, 2023-03-07:
@ZIMkaRU I fix this issue. Could you please try it with the attached binary and confirm?

[kaluma-rp2-pico-w-1.1.0-beta.1_test_0306.uf2.zip](https://github.com/kaluma-project/kaluma/files/10906492/kaluma-rp2-pico-w-1.1.0-beta.1_test_0306.uf2.zip)

communix, 2023-03-11:
@ZIMkaRU Thank you for your confirmation. And sorry, I found a critical bug it this FW. the network driver does not work with this code change. I'll fix this issue and share the other FW which can resolve this issue and network driver issue.

communix, 2023-03-12:
@ZIMkaRU I fixed the issue and we can use PIO and network driver with the attached FW.
Please re-open this issue if you have the issue.

[kaluma-rp2-pico-w-1.1.0-beta.1_test_0311.uf2.zip](https://github.com/kaluma-project/kaluma/files/10949749/kaluma-rp2-pico-w-1.1.0-beta.1_test_0311.uf2.zip)

closed 2023-03-12 by communix