kaluma-project/kaluma #560: kaluma flash alternative

realJoshByrnes, 2023-03-08:
I'm having a little issue using `kaluma flash` - https://github.com/wokwi/rp2040js/issues/123 (seems to be a wokwi bug) but it got me thinking...

* `kaluma put` works as expected, but there's no way to `put` the file into the flash memory. As it works differently to flash (not using ymodem) it would be nice to be able to write to the flash.
* `kaluma flash` (should) writes the file to the device, but there's no way to generate a firmware.uf2 file containing the script, which can then be moved to another device)

Both of these would allow development locally (without using actual hardware) while the issue above exists,  and I think they'd both be great features to have (particularly the second, so the resulting image can be shared).

Sorry if I've missed something and either of these features already exist. I did poke around in the source and was unable to find a suitable alternative.

communix, 2023-03-11:
I think you want to have uploading one file which have FW, files in file system and java script source code. But we don't have a plan to support this feature now. Since it's a open source project, we are welcome for you to implement this feature for this project.

jakeg, 2023-09-25:
Seems similar to my issue https://github.com/kaluma-project/kaluma-cli/issues/38 

I expect I'll read the source later, but I'm scratching my head as to where the usercode is stored. If I reflash the .uf2 file (even after switching to MicroPython .uf2) my usercode is still there.