kaluma-project/kaluma #562: Http Support for CYW43 Module

AcousticTypewriter, 2023-03-23:
Hey there,

Thanks for the great work on this project. Kaluma is fantastic.

I've been trying to implement contacting a server with a Pico W, and  I couldn't figure out why it wasn't working until I found that the docs say that there's only TCP implementation for the CYW43 network device driver. I was wondering if HTTP support is part of the roadmap or what documents I should reference for implementing support for it

Thanks so much

communix, 2023-04-08:
@AcousticTypewriter Thank you for using Kaluma.
The basic function of the http server is working now but it has some issue now. I'm trying to fix the issue. Please check the update in #557. you can find the test FW in there. I'll let you know once I fix the http issues.

communix, 2023-06-25:
Duplicated issue #570

closed 2023-06-25 by communix