kaluma-project/kaluma #569: IRQ not working

FlorianRueb, 2023-04-16:
Hi there!

I don't get the GPIO.irq function working. When I try to use the code in the example (using PIN 26 and 27 for Input / Output), I get plenty of lines with "Event undefined is triggered on the GPIOundefined" and the code stops to work. 

I also tried to code a small IRQ-driven function, but the code stops after the first IRQ event and I don't know where the problem is based on. Maybe you have any hint for me? I am using the Seeeduino XIAO Board where Input 0 and 1 is connected to GPIO26/27

Here's my code:
```
// Input Switch zwischen Masse und PIN26 (D0) > schaltet das blinkende NeoPixel an bzw. aus

// Setup Variables
let isPixelValue = 0; // Default NeoPixel RGB Value (Off)
let blinkOnOff = 1;

// Setup for NeoPixel
const {NeoPixel} = require('neopixel');
const neoPixel = new NeoPixel(12, 1); // 1 pixels on GPIO12
const npPower = 11; // XIAO NeoPixel Power Source at RP2040 / GPIO11

const senseOne = 26; // XIAO RP2040: PIN A0/D0/P26
const poutput27 = 27; // XIAO RP2040: PIN A1/D1/P27

// Setup Inputs / Outputs / GPIOs
const { GPIO } = require('gpio');
const pinput = new GPIO(26, INPUT_PULLUP); // GPIO1 is input port which check the state change
pinMode(npPower, OUTPUT); // XIAO NeoPixel Power Source at RP2040 / GPIO11

// Start by turning the light ON (npPower)
digitalWrite(npPower, HIGH); // Power on: NeoPixel

//Callback Function for Blinking LED
function npToggle(){
    isPixelValue = neoPixel.getPixel(0);
    if(isPixelValue == 0 && blinkOnOff == 1){
        console.log('blinOnOff: ' + blinkOnOff);
        neoPixel.clear();
        console.log('{"Pixel": ' + isPixelValue + '}\n');
        neoPixel.setPixel(0, neoPixel.color(255, 52, 179)); // LED-Color of Pixel 1
        neoPixel.show();
    }
    else if(isPixelValue > 0 && blinkOnOff == 1){
        console.log('{"Pixel": ' + isPixelValue + '}\n');
        console.log('blinOnOff: ' + blinkOnOff);
        neoPixel.setPixel(0, neoPixel.color(0, 0, 0)); // Pixel 1 aus
        neoPixel.show();
    }
    else if(blinkOnOff == 0){
        console.log('blinOnOff: NULL' + blinkOnOff);
    }
}

//Callback Function for switching LED >OFF<
function npSwitchOff() {
    console.log('{"Pixel": "AUS"}\n');
    neoPixel.clear();
    neoPixel.show();
    blinkOnOff = 0;
    delay(50);
    return;
}

//Callback Function for switching LED >OFF<
function npSwitchOn() {
    blinkOnOff = 1 - blinkOnOff;
    console.log('{"blinkOnOff": ' + blinkOnOff + '}\n');
    delay(50);
    return;
}

pinput.irq(npSwitchOn, CHANGE);

setInterval(() => {
    npToggle();
}, 1000);
```
Thanks,
Florian

communix, 2023-04-22:
Let me check it and update it.

communix, 2023-04-30:
@FlorianRueb I fixed this issue. Could you please try it with this FW?

[kaluma-rp2-pico-1.1.0-beta.1_test_0429.uf2.zip](https://github.com/kaluma-project/kaluma/files/11360709/kaluma-rp2-pico-1.1.0-beta.1_test_0429.uf2.zip)

I verified it with this code.
```
// Set interrupt on the GPIO 26
const { GPIO } = require('gpio');
// GPIO 26 and GPIO 27 are connected together
const pin0 = new GPIO(26, INPUT); // GPIO 26 is input port which check the state change of GPIO 27
const pin1 = new GPIO(27, OUTPUT); // GPIO 27 is output
pin1.low(); // Set LOW on the GPIO 27

// Callback function for the GPIO 26
function callback0(pin, mode) {
  console.log('Event ' + mode + ' is triggered on the GPIO' + pin);
}

// Set interrupt callback function on GPIO 26
pin0.irq(callback0, CHANGE);

pin1.high(); // Callback is called because GPIO26 is changed from LOW to HIGH
delay(1);
pin1.low(); // Callback is called because GPIO26 is changed from HIGH to LOW

```

FlorianRueb, 2023-05-01:
Hi Jay,

yes, it now seems to work properly! Thank you!

It seems to me that it is not possible to set two different IRQ Events on the same pin, right? I wanted to set up two different callback functions in case the input pin is RISING or FALLING. Setting up two different callbacks ends up in a stop.

communix, 2023-05-02:
@FlorianRueb Yes, we support only one callback for the one pin but I don't know why you need two difference callback function for the RISING and FALLING. the call back function has the pin and mode value and you can distinguish the RISING and FALLING using mode value in the callback function. So you don't need to have separate two callback functions for one pin. you can implement two different RISING and FALLING process in the one callback function.

closed 2023-05-02 by communix