kaluma-project/kaluma #570: HTTP Module not working on Raspberry Pi Pico W

n0bisuke, 2023-04-17:
Hi there

I'm trying the HTTP module on Raspberry Pi Pico W, but it doesn't work. There doesn't seem to be any particular error.

I tried the following code.

```javascript
const { WiFi } = require('wifi');
const wifi = new WiFi();
const http = require('http');

wifi.scan((err, scanResults) => {
    if (err) {
        console.error(err);
    } else {
        wifi.connect({ ssid: 'SSID', password: 'PASS' }, (err) => {
            if (err) {
                console.error(err);
            } else {
                console.log(`-----Wi-Fi connected-----`);        
                try {
                    http.get({ host: 'hoge.hoge.com', path: '/' }, function (response) {
                        response.on('data', function (chunk) {
                            console.log('received: ', chunk);
                        });
                    });
                } catch (error) {
                    console.log(`---`);
                    console.log(error);
                }
            }
      });
    }
});
```

It seems that the people commenting in this discussion are having the same issue.
> https://github.com/kaluma-project/kaluma/discussions/550 

Anyone have any ideas?

communix, 2023-04-22:
@n0bisuke Could you try it with this test FW?

[kaluma-rp2-pico-w-1.1.0-beta.1_http_test_0409.uf2.zip](https://github.com/kaluma-project/kaluma/files/11188052/kaluma-rp2-pico-w-1.1.0-beta.1_http_test_0409.uf2.zip)

AcousticTypewriter, 2023-04-24:
@communix I tried it with that FW and it did not work or give an error. If I wrapped the http.get in a setInterval with a console log in it, I got the error, "Error: Remote I/O error" and lots of "[CYW43] got unexpected packet -9"

Any ideas?

communix, 2023-04-30:
@AcousticTypewriter Let me check it and let you know.

anubhavgupta, 2023-05-26:
Thanks for all your work @communix.
I am also facing the same issue, tried with the latest firmware as well.

anubhavgupta, 2023-05-26:
```
 const d = http.get({
      host: '160.153.63.10', // x.com, a simple website
     path: '',
     }, function (response) 
          response.on('data', function (chunk) {
          console.log('received: ', chunk);
        });
     });
```

Ok, the above code worked, though I am not able to put the URls directly, I have to put the ip address of the websites.(Some/most of the websites doesn't allow accessing the website directly via ipaddress)

communix, 2023-05-27:
@anubhavgupta Thank you. Good finding. I'll check that point.

communix, 2023-05-28:
@anubhavgupta Thanks to your help I found out that current network driver does not support URLs. Let me implement URL feature in the pico-w network driver.

communix, 2023-05-29:
@anubhavgupta @AcousticTypewriter I implement URL feature in the pico-w network driver. Could you please check this issue with new FW?
[kaluma-rp2-pico-w-1.1.0-beta.2_URL_test_0528.uf2.zip](https://github.com/kaluma-project/kaluma/files/11588519/kaluma-rp2-pico-w-1.1.0-beta.2_URL_test_0528.uf2.zip)

AcousticTypewriter, 2023-05-29:
@communix No luck. Fails silently, even with ip addresses. Not sure if this is my code or firmware. Could you give me example code to run on my device?

anubhavgupta, 2023-05-29:
Let me test and get back.

communix, 2023-05-29:
@AcousticTypewriter Hi I tested with the code in [this issue ](https://github.com/kaluma-project/kaluma/issues/570#issue-1671242822) and the host is 'httpbin.org'. If it's now work with IP address, the new FW does not help because I just implement URL support feature in the new FW.

anubhavgupta, 2023-05-29:
It seems to be working for me(though seems to be some issue with the request).

I have tried it with x.com, minwiz.com ( <10kb websites )

But instead of returning the actual content it retured:
```
'<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 4.01 Transitional//EN" "http://www.w3.org/TR/html4/loose.dtd">
<HTML><HEAD><META HTTP-EQUIV="Content-Type" CONTENT="text/html; charset=iso-8859-1">
<TITLE>ERROR: The request could not be satisfied</TITLE>
</HEAD><BODY>
<H1>400 ERROR</H1>
<H2>The request could not be satisfied.</H2>
<HR noshade size="1px">
Bad request.
We can't connect to the server for this app or website at this time. There might be too much traffic or a configuration error. Try again later, or contact the app or website owner.
<BR clear="all">
If you provide content to customers through CloudFront, you can find steps to troubleshoot and help prevent this error by reviewing the CloudFront documentation.
<BR clear="all">
<HR noshade size="1px">
<PRE> 
```

anubhavgupta, 2023-05-29:
Here is my (quick and hacky) code:

```
const { PicoCYW43, PicoCYW43WIFI, PicoCYW43Network } = require('pico_cyw43');
global.__netdev  = new PicoCYW43Network();
global.__ieee80211dev = new PicoCYW43WIFI();

var http = require('http');
var { WiFi } = require('wifi');
var wifi = new WiFi();
function connectToWifi() {

    wifi.reset(()=>{

        wifi.scan((err, scanResults) => {
            if (err) {
              console.error(err);
            } else {
              console.log(scanResults);
                wifi.connect({ ssid: '<SSID>', password: '<PASS>' }, (err) => {
                    if (err) {
                        console.error(err);
                    } else {
                        console.log("Connected");
                        try{
                            const d = http.get({
                                 host: 'minwiz.com', // x.com
                                 path: '/',
                              }, function (response) {
                                console.log('connected 6');
                                response.on('data', function (chunk) {
                                    console.log('received: ', chunk);
                                });
                            });
                            globalThis.d = d; // later log `d._parser.body` from shell to see actual text content
                        } catch(ex){
                            console.log(ex);
                        }
                    }
                });
            }
          });

    })
    
}

setTimeout(()=>{
    //console.log(http);
    connectToWifi();
}, 5000);

```

communix, 2023-05-29:
> It seems to be working for me(though seems to be some issue with the request).
>
> I have tried it with x.com, minwiz.com ( <10kb websites )
>
> But instead of returning the actual content it retured: [...]

@anubhavgupta Thank you, The result is same when you try it with IP address? or It's expected response with IP address?

anubhavgupta, 2023-05-29:
same result, when i try with ip address.

communix, 2023-05-29:
@anubhavgupta Thank you let me check that point!!!

communix, 2023-06-25:
@AcousticTypewriter @anubhavgupta I finally found the issue in the http code. please try it with the attached test FW and let me know your test result.
[kaluma-rp2-pico-w-1.1.0-beta.2_0624_http_get_test.uf2.zip](https://github.com/kaluma-project/kaluma/files/11858643/kaluma-rp2-pico-w-1.1.0-beta.2_0624_http_get_test.uf2.zip)

I tested with this test code. The host in the http header is needed for the "httpbin.org".
```
const { WiFi } = require('wifi');
const wifi = new WiFi();
const http = require('http');

wifi.scan((err, scanResults) => {
  if (err) {
      console.error(err);
  } else {
    wifi.connect({ ssid: 'myssid', password: 'mypassword' }, (err) => {
      if (err) {
          console.error(err);
      } else {
          console.log(`-----Wi-Fi connected-----`);
      }
    });
  }
});

const options = {
  host: 'httpbin.org',
  port: 80,
  path: '/get',
  method: 'GET',
  headers: {Host: 'httpbin.org', Accept: 'application/json'}
};

const req = http.request(options, (res) => {
  console.log(`STATUS: ${res.statusCode}`);
  console.log(`HEADERS: ${JSON.stringify(res.headers)}`);
  res.on('data', (chunk) => {
    console.log(`BODY: ${chunk}`);
  });
  res.on('end', () => {
    console.log('No more data in response.');
  });
});

req.on('error', (e) => {
  console.error(`problem with request: ${e.message}`);
});

// Finish the request
req.end();
```

anubhavgupta, 2023-06-25:
Thanks @communix 

When I tried the shared F/W with the following config, it gave me below error:

Config: 
``` 
const options = {
                            host: 'minwiz.com',
                            port: 80,
                            path: '/',
                            method: 'GET',
                            headers: {
                                Host: 'minwiz.com',
                                Accept: 'text/html'
                            }
                         };
```

Error: 
```
'<html>
<head><title>301 Moved Permanently</title></head>
<body>
<center><h1>301 Moved Permanently</h1></center>
<hr><center>CloudFront</center>
</body>
</html>
'
```
Not sure, if it is not working because this website seems to accept https only?

communix, 2023-06-25:
@anubhavgupta I check it with curl command and it's the same.
Please check the HTTP 301 error, https://en.wikipedia.org/wiki/HTTP_301

```
user@dev-laptop ~ $ curl -X GET "http://minwiz.com/get"
<html>
<head><title>301 Moved Permanently</title></head>
<body>
<center><h1>301 Moved Permanently</h1></center>
<hr><center>CloudFront</center>
</body>
</html>
```

dreanmer, 2023-06-27:
@communix your last firmware worked few times, but it's crashing on most of the times when connecting with the wifi (this isn't happening on beta 1.1 release)

communix, 2023-07-03:
All the HTTP issues are fixed except #593. So we need to use IP address for the HTTP protocol now. So I close this issue.

closed 2023-07-03 by communix