kaluma-project/kaluma #573: Potential memory leak in `spi.transfer`

alexander-daniel, 2023-05-04:
Hello! I seem to have encountered a potential memory leak in kaluma `spi.transfer` method call.

I have tried both `1.0.0` and `1.1.0-beta.1` firmware.

`1.0.0` slowly creeps up memory usage as i call `spi.transfer` in a loop. (10ms, 100ms, 250ms, 1ms intervals)

`1.1.0-beta.1` seems to crash much quicker -- and I get a `Panic Out of Memory` (But I do not get this message on `1.0.0`.)

I've attached the code I'm using, I was trying to write a little library for the MCP3008. It's almost there, but I ran into this during testing, where the Pico would lock up after a given time (made quicker by faster loops). All the code works just fine if you comment out any `spi.transfer` call, and it never panics or runs out of memory or locks up.

Weirdly enough, the crash is so bad I have to reset the pico (totally reflash a `nuke-flash.uf2` and then put `kaluma` back on.) This kinda makes the debugging process a bit slow, but I'm 99% confident in what I've stated above :) 

LOVING kaluma! Really great work, hoping I can contribute back here. Thank you so much

### `index.js`
```js
const MCP3008_CHANNEL_COUNT = 8;

class MCP3008 {
  constructor(spi, cs) {
    this.spi = spi; // instance of Kaluma SPI
    this.cs = cs; // chip select pin
  }

  read(channel) {
    if (channel < 0 || channel >= MCP3008_CHANNEL_COUNT) {
      throw new Error(`Invalid channel: ${channel}`);
    }

    const txBuffer = new Uint8Array([
      0b00000001, // Start bit
      (8 + channel) << 4, // Single-ended channel select
      0 // Dummy byte to clock in the data
    ]);

    // Select the device  
    digitalWrite(this.cs, LOW);

    // Transfer the txBuffer to the device, and read the rxBuffer back in from the device
    const rxBuffer = this.spi.transfer(txBuffer);
    const value = ((rxBuffer[1] & 0x03) << 8) + rxBuffer[2];
  
    // Deselect the device
    digitalWrite(this.cs, HIGH);

    return value;
  }
}

module.exports = MCP3008;
```

### `example.js`
```js
const { SPI } = require('spi');
const MCP3008 = require('./index.js'); // replace this with `mcp3008` in your code and package.json.

const SPI_BUS = 0;
const CLOCK_SELECT_PIN = 17;

// Setup our chip select pin for the mcp3008
// we have to set it to output, and set default to HIGH.
// this pin will be pulled down to LOW when we want to read 
// from the MCP3008 peripheral.
pinMode(17, OUTPUT);
digitalWrite(17, HIGH);

// Setup our SPI bus that we'll use to communicate with the MCP3008
const spi = new SPI(SPI_BUS);
const mcp3008 = new MCP3008(spi, CLOCK_SELECT_PIN);

let value;

function loop() {

  // this read function will automatically set the state of the provided
  // chip select pin so it can communicate to the mcp3008 peripheral.
  // all you have to provide is the channel you wish to read (0-7)
  value = mcp3008.read(1);
  console.log(value);

  setTimeout(loop, 10);
}

loop();
```

alexander-daniel, 2023-05-04:
I've also tried casting less variables and assigning directly to class instances, to see if there was something there, but that didn't do much (or rather didn't find the source of the leak)

communix, 2023-05-06:
@alexander-daniel version 1.0.0 has a memory leakage issue but I think I fixed the memory leakage issue in version 1.1.0-beta.1. Let me check this issue and let you know.

For the debugging, you don't have to download FW when you have an lockup issue. please check the below link. You can erase flash or re-write new code after skipping the code running. 
https://kalumajs.org/docs/boards/pico#skip-code-loading-on-boot

alexander-daniel, 2023-05-06:
Oh wonderful, I'll give that skip code loading a shot! 

Thank you for taking a look! I'll try again tonight with fresh eyes too.

communix, 2023-05-07:
@alexander-daniel I don't think it's a memory leakage issue. I run SPI functions and check the memory, I can't see any memory leakage. But when I try to run your code, I can see the system lockup issue at malloc() and free() function which is the function of pico-sdk. Let me check how we solve this issue.

alexander-daniel, 2023-05-07:
Sorry for my inexperience, but what do you mean you see the system lockup issue? It's not related to memory? Thank you so much for taking a peek, happy to help test or do anything to help! :D

alexander-daniel, 2023-05-07:
And yes, that makes sense. `1.1.0-beta.1` didn't have the same behaviour, but still locked up, with that additional message. I just assumed it was the same thing, but to confirm the two firmwares DID have different behaviour

communix, 2023-05-07:
@alexander-daniel I think I fix this issue. Please pull the master branch and build new image with it. If you don't have a build env. please let me know I'll release test FW for you. (please let me know which board you are using, pico? or pico-w?)

alexander-daniel, 2023-05-08:
@communix Just tested it and it works! No more panic, no more lockups! I really appreciate you taking a peek and getting a fix out so fast. I'll share my libraries with you now that I've got SPI working!

For my understanding, I took a look at the PR but I don't really understand what's happening. If you have time, would you be able to elaborate a bit on what the issue actually was?

communix, 2023-05-14:
@alexander-daniel Thank you for your confirmation, I'll release new beta version with this change next Sunday.
The issue is double free the buffer. we can create arraybuffer using` jerry_create_arraybuffer_external(length, buf, buffer_free_cb);` and the `buf` is freed in the `buffer_free_cb` function. So we don't need to free the `buf`. So I remove the free(buf).

closed 2023-05-14 by communix

alexander-daniel, 2023-05-20:
That makes sense! Thank you for taking the time to explain and fix this issue. It's been a great experience diving into this. Keep up the awesome work, it's super appreciated!