kaluma-project/kaluma #585: GPIO IRQs are interrupting running JS code

skadisch, 2023-06-20:
GPIO IRQs registered via "gpio.irq(callback[, event])" are executed as they arrive, interrupting the currently running program. Normally, JavaScript environments are of cooperative nature and schedule events in the event loop where they will be executed when the current task completes.

Further, using IRQs leads to a complete crash / hang, even in simple use cases. Most likely as things are not reentrant / thread safe.

Ignoring that this currently crashes, is it expected to behave like this? If yes, it should be mentioned in the documentation, also pointing out potential risks. If not, I would be happy to provide a PR.

If it is about speed, than at least it should run in an isolated JS context and communicate via messages, like a JS worker.

communix, 2023-06-25:
@skadisch Thank you for using Kaluma. If you can see the crash/hang when you use IRQs, could you please share your code?

communix, 2023-06-25:
@skadisch Which version did you use for the test? Please use the latest FW. Old FWs has IRQ issue.
https://github.com/kaluma-project/kaluma/releases/tag/1.1.0-beta.2