kaluma-project/kaluma #586: calling a function is slow & rtc cant detect correct time intervals

madmagic007, 2023-06-20:
minimal code example:

```javascript
function func() {
    for (let i = 0; i < 8192; i++) {
        test();
    }
    console.log("after");
}

function test() {}

setInterval(func, 0);```

I tried on both pico and pico w boards, the "after" only gets printed like 5 times per seconds in the console.
if i remove or comment out the `test();`, then it`s as fast as the console can handle prints it seems.
so its not the looping, its the calling a function inside a loop that seems to be very slow.
if i define the `func` inside the setInterval or not, its the same result.

i tried using all version for both boards, same result.

Plus, using the RTC (or any timing library), seems to be only able to to detect 1000ms delays?
eg if i put `rtc.getTime();` before and after the for loop and compare them, its either 0 or 1000, nothing in between.

communix, 2023-06-25:
@madmagic007
Sorry, Yes it's slower than C code but I don't think I can make it faster. It's the java script engine and rp2040 performance issue.

The RTC is very slow clock so you can't detect ms delay. Kaluma use the default RTC timer in RP2040 and it's 1Hz clock. So you can't use RTC to detect fast timing (less than 1sec).
Instead of the RTC, Kaluma provide millis() and micros() function. please find the details in below link
https://kalumajs.org/docs/api/timers

communix, 2023-07-03:
Close this issue. Please re-open it if anyone has question about this topic.

closed 2023-07-03 by communix