kaluma-project/kaluma #587: TypeScript Support?

leventkaragol, 2023-06-25:
Any plans on typescript support? I'm not talking about directly loading typescript files with flash. But at least some type support like @types/kaluma would be nice

pclokcer, 2023-07-31:
if we build typescript to javascript, I think it will work ? @communix

communix, 2023-08-06:
@pclokcer Yes. it works and you can publish it to public (https://kalumajs.org/packages/)  and other people can use it.

samboylett, 2023-10-15:
We don't need JerryScript to support it. Same way we transpile TypeScript to JavaScript for web we can do for kaluma. Really all we need is a declaration file for the in built modules/constants. I've been building one up myself but it would be good to have an official one. 

Actually looking at the source code all of this is already in javascript, so if we can convert it to typescript we can get declaration files for it (and transpile back to JS for JerryScript to use)

jt000, 2023-12-09:
@samboylett, I started the process of creating a type definition for Kaluma in [DefinitelyTyped](https://github.com/DefinitelyTyped/DefinitelyTyped/pull/67850). Still quite a few components left to do, but got a good start on many of them...

jt000, 2023-12-13:
@niklauslee or @communix, would you be interested in co-owning the typescript definition files in DefinitelyTyped with me? https://github.com/DefinitelyTyped/DefinitelyTyped/pull/67850#discussion_r1423149517

niklauslee, 2023-12-14:
@jt000 Wow this is amazing contribution👍. I'm difficult to contribute or maintain.😥

jt000, 2023-12-31:
@leventkaragol, I've been using typescript\webpack with Kaluma pretty successfully with the solution provided in https://github.com/jt000/kaluma-typescript-demo... Does this satisfy your typescript needs?

leventkaragol, 2024-01-01:
Thanks, I'll take a look

edceds, 2025-08-16:
using ts + @types/kaluma fine on https://github.com/edceds/pi-pico-kaluma-ts