kaluma-project/kaluma #590: running the shell command related issue

faroha-111, 2023-06-30:
i successfully ran this command "kaluma shell" on the command prompt and received the welcome message, but now when i try to run it again this shows "port not found: @2e8a". the CMD was run as an administrator in both times.

communix, 2023-07-01:
@faroha-111 It seems to be USB port issue. Could you please check device manager (If you use Windows OS) and see the USB device driver works well? Kaluma use pico USB serial interface. So please check this point.

And you can use "Kaluma shell --port COM3" to set the com port, Please see this document, https://github.com/kaluma-project/kaluma-cli#shell-command

[image]

communix, 2023-07-02:
@faroha-111 This issue may be related to the javascript code issue in the flash. So please skip the code in the flash.
This is the link of the skipping code loading on boot.
https://kalumajs.org/docs/boards/pico#skip-code-loading-on-boot

faroha-111, 2023-07-02:
It worked thank you so much, but now I have a new problem. When I ran the ".flash -w index.js" command, it started typing a lot of 'C's, so I waited for hours and then it stopped Cing without sending any finishing message, so I aborted it and flashed the code again, and the same things happened again. What can I do?
Thanks for your concern.

[image]<https://user-images.githubusercontent.com/44126592/250239082-0cef342f-9468-4d9d-b2a8-0100cc0e5233.png>

faroha-111, 2023-07-02:
I repeated the same steps in addition to the instructions in the last link you asked me to check, and it has been typing C's for over 3 hours so far.
I am sorry for being that annoying but It turned out that I am kind of a beginner, also my competition's deadline is not far so I'm nervous. Thank you again.

https://kalumajs.org/docs/boards/pico#skip-code-loading-on-boot

communix, 2023-07-02:
@faroha-111
.flash -w command is manual YMODEM code download feature. So you need to manually run YMODEM file transfer function.
Kaluma CLI provide easy code download feature. Please use Kaluma CLI to download your code.
```
> kaluma flash index.js
```
https://kalumajs.org/docs/getting-started#flash-your-code

closed 2023-08-13 by communix