kaluma-project/kaluma #596: Error is code example provided on Documentation page for HTTP

prithwirajbose, 2023-07-07:
I am referring the documentation page for HTTP here: [https://kalumajs.org/docs/api/http](https://kalumajs.org/docs/api/http)

In the http.createServer([requestListener]) section there is small syntax error. It might be very confusing for someone who is starting out new and doesn't know how to debug. Because this syntax error will cause the program to not work when flashed to Pico W.

See in the screenshot below
[image]

Instead of `response` the variable name should be `res`. I know this is trivial but I spent some good amount of time overlooking this simple bug and was just about to try back Micropython instead of Kaluma, loosing hope :)

communix, 2023-07-10:
@prithwirajbose Thank you. I'll fix the examples in the documents.

communix, 2023-09-10:
This issue is fixed.

closed 2023-09-10 by communix