kaluma-project/kaluma #598: reboot problem

pclokcer, 2023-08-05:
I want to reboot my board. how can I reboot?

communix, 2023-08-06:
@pclokcer To reboot the system, plug out the power and plug in the power. If you need to skip the javascript code, please see this document (https://kalumajs.org/docs/boards/pico#skip-code-loading-on-boot)

pclokcer, 2023-08-06:
in fact, I want to do watchdog

communix, 2023-08-13:
@pclokcer OK. I got it. Let me think about how we can implement watchdog.

communix, 2023-09-04:
@pclokcer watchdog class is implemented in the master branch. Please try it.

This is the sample code.
```
const { WDT } = require('wdt');
const wdt = new WDT(1000); // 1 sec watchdog timeout

var timer = 0;

var timer_id = setInterval(()=> {
    wdt.feed(); // Reset watchdog timer
    timer += 500;
    console.log("Reset watchdog timer " + timer + " ms");
    if (timer > 10000) { // Do not reset after 10 sec.
        clearInterval(timer_id);
        console.log("Watchdog reset after 1 sec.");
    }
}, 500);
```

pclokcer, 2023-09-04:
vauv thank you

closed 2023-09-04 by pclokcer