kaluma-project/kaluma #624: Onboard LED not blinking in Pico W

kashyap1113, 2023-10-17:
I am trying to blink LED using Quick Start guide code. But its not working because in Pico W, the LED is not connected to pin 25. It is connected to WL_GPIO0 on the Infineon 43439 wireless chip for Pico W.

[Check the official docs](https://www.raspberrypi.com/documentation/microcontrollers/c_sdk.html#blink-an-led)

I tried to replace 25 with 'WL_GPIO0' in code as shown below but it didn't work:
```
// index.js
const led = 'WL_GPIO0';
pinMode(led, OUTPUT);
setInterval(() => {
  digitalToggle(led);
}, 1000);
```

For reference, following is the equivalent python code that works with micropython
```
from machine import Pin
from time import sleep

led = Pin('WL_GPIO0', Pin.OUT)
while True:
    led.toggle()
    sleep(1)
```

jpacora, 2023-10-19:
Use this code (as shown in https://kalumajs.org/docs/boards/pico-w):

```
const { PicoCYW43 } = require('pico_cyw43');
const pico_cyw43 = new PicoCYW43();

// Blink on-board LED
setInterval(() => {
  if (pico_cyw43.getGpio(0)) {
    pico_cyw43.putGpio(0, false); // turn-off LED
  } else {
    pico_cyw43.putGpio(0, true); // turn-on LED
  }
}, 1000);
```

kashyap1113, 2023-10-19:
My sincere apologies for creating an unnecessary issue. I'm unsure how I overlooked the documentation section.

closed 2023-10-19 by kashyap1113