kaluma-project/kaluma #628: Incomplete TCP packet when sending Uint8Array packet

jpacora, 2023-12-05:
I'm receiving incomplete packet on my TCP server when sending Uint8Array packet. The pico send ```<Buffer 10 16>``` but the same code on my mac send ```<Buffer 10 16 00 04 4d 51 54 54 04 02 00 3c 00 0a 4d 51 54 54 5f 33 33 31 32 32>```

```const { WiFi } = require('wifi')
const wifi = new WiFi()
const net = require('net');

wifi.connect({
  ssid: 'HIDDEN',
  password: 'HIDDEN'
}, (err) => {
  if (err) {
      console.error(err)
  } else {
     
    var client = net.createConnection({
      host: "8.tcp.ngrok.io",
      port: 15504,
    }, () => {
      const packet = new Uint8Array([16,22,0,4,77,81,84,84,4,2,0,60,0,10,77,81,84,84,95,51,51,49,50,50])
      // 'connect' listener.
      console.log('connected to server!');
      client.write(packet);
    });
    client.on('data', (data) => {
      console.log(data);
      client.end();
    });
    client.on('end', () => {
      console.log('disconnected from server');
    });

  }
})```

communix, 2023-12-11:
@jpacora Thank you for your issue report. I think I fix this issue. Could you please build new FW with this patch and confirm it's fixed?
https://github.com/kaluma-project/kaluma/commit/5d3a844e35ad3d864258eb877062b918542019d8

jpacora, 2023-12-11:
Now its working but I'm having another issue, the "data" event of the client socket is always string instead of the Uint8Array.

```
const { WiFi } = require('wifi')
const wifi = new WiFi()
const net = require('net')

wifi.connect({
  ssid: 'HIDDEN',
  password: 'HIDDEN'
}, (err) => {
  if (err) {
      console.error(err)
  } else {
     
    var client = net.createConnection({
      host: "52.29.220.148",
      port: 1883,
    }, () => {
      const packet = new Uint8Array([16,22,0,4,77,81,84,84,4,2,0,60,0,10,77,81,84,84,95,51,51,49,50,50])
      // 'connect' listener.
      console.log('connected to server!');
      client.write(packet);
    });
    client.on('data', (data) => {
      console.log("---- onData ----")
      console.log(typeof data)
      console.log(data);
    });
    client.on('end', () => {
      console.log('disconnected from server');
    });

  }
})
```
On the pico this returns:
```
---- onData ----
string
 
```

But on my mac:
```
---- onData ----
object
<Buffer 20 02 00 00>
```

communix, 2023-12-12:
@jpacora Thank you let me fix it as well.

communix, 2023-12-17:
@jpacora I fix the issue, Could you please try it with the latest code in this branch, https://github.com/kaluma-project/kaluma/tree/develop/tcp_uint8array_write_bug ?

communix, 2023-12-18:
@jpacora Sorry this fix has side-effect. I'll update it soon.

communix, 2023-12-18:
@jpacora  I fix the issue and side-effect, Could you please try it with the latest code in this branch, https://github.com/kaluma-project/kaluma/tree/develop/tcp_uint8array_write_bug ?

jpacora, 2023-12-19:
Nice, now is working ;)

closed 2023-12-21 by communix