kaluma-project/kaluma #630: pulseRead does not work with DHT

ZIMkaRU, 2023-12-21:
Looks like `pulseRead` does not work with `DHT22`

---

checked with this lib https://github.com/niklauslee/dht
made simple example:

```js
const { DHT } = require('dht')

const pin = 15
const dht = new DHT(pin, DHT.DHT22)

setInterval(() => {
  const res = dht.read()
  console.log('[DHT res]:', res)
  console.log(`Temp: ${dht.temperature} C`)
  console.log(`Humidity: ${dht.humidity} %`)
}, 5000)
```

and `pulseRead` always returns `null`: https://github.com/niklauslee/dht/blob/main/index.js#L74
I connected an oscilloscope to the signal pin of the sensor to make sure it works
trigger pulses are processed successfully and the sensor returns a data packet
but it looks like reading is not happening

[image]

[image]

[image]

[image]

---

I think I'm not the only one having this problem
https://github.com/niklauslee/dht/issues/2

ZIMkaRU, 2023-12-21:
**Update**: the issue is in the trigger section of the `pulseRead`
when the last trigger impulse is set to down then input is pulled up
I was able to fetch bits when added a third trigger value of `20`

```js
var bits = pulseRead(this.pin, 100, {
  timeout: 25000,
  startState: LOW,
  mode: INPUT,
  trigger: {
    startState: HIGH,
    interval: [10000, 18000, 20],
  },
});
```

https://github.com/niklauslee/dht/blob/main/index.js#L80

ZIMkaRU, 2023-12-22:
@communix several thoughts:
maybe when we set the `INPUT` mode after the trigger
https://github.com/kaluma-project/kaluma/blob/c3071d6868a000607349aa29126f5a32c02c0f19/src/global.c#L276
we should set `false` for `gpio_set_dir(pin, false);`
before setting `gpio_set_input_enabled(pin, true);`
https://github.com/kaluma-project/kaluma/blob/c3071d6868a000607349aa29126f5a32c02c0f19/targets/rp2/src/gpio.c#L73
in order to disable `OUTPUT` mode before setting `INPUT` one
https://github.com/kaluma-project/kaluma/blob/c3071d6868a000607349aa29126f5a32c02c0f19/targets/rp2/src/gpio.c#L66

ZIMkaRU, 2023-12-22:
I made this fix: https://github.com/kaluma-project/kaluma/pull/631
and built a beta release on my own: 
[kaluma-rp2-pico-1.1.0-beta.4.zip](https://github.com/kaluma-project/kaluma/files/13753637/kaluma-rp2-pico-1.1.0-beta.4.zip)
works good

communix, 2023-12-29:
@ZIMkaRU Thank you for your contribution. Your code looks good and it's merged.
Thanks to your code, I have a chance to review GPIO input/output code and add more code to fix this issue.
Please try it with the latest code in the master branch.

ZIMkaRU, 2023-12-30:
@communix I've built and tested many versions of the combinations for `gpio_set_dir`/`gpio_set_input_enabled` and it works only if remove `gpio_set_input_enabled(pin, false)` which is placed after `gpio_set_dir(pin, GPIO_OUT)` for `KM_GPIO_IO_MODE_OUTPUT` mode
https://github.com/kaluma-project/kaluma/blob/master/targets/rp2/src/gpio.c#L67

communix, 2023-12-30:
@ZIMkaRU Thank you. Let me remove gpio_set_input_enabled(pin, false)

ZIMkaRU, 2023-12-31:
Sure, thanks

ZIMkaRU, 2023-12-31:
checked, works well

closed 2023-12-31 by ZIMkaRU