kaluma-project/kaluma #646: Support for new RP2350 (Pi Pico 2)

jakeg, 2024-08-29:
Any idea if Kaluma will just work with the new microcontroller, or does it need some code changes?

communix, 2024-10-28:
@jakegI apologize for the late reply. I'm hesitant to commit to supporting this project, as it seems to be inactive. However, if you and others are actively interested in using the Kamula project for the RP2350 board, I'd be happy to consider dedicating time to explore this further.

communix, 2024-10-28:
@jakeg I apologize for the late reply. I'm hesitant to commit to supporting this project, as it seems to be inactive. However, if you and others are actively interested in using the Kamula project for the RP2350 board, I'd be happy to consider dedicating time to explore this further.

jakeg, 2024-10-29:
I teach computer science at a school in the UK, and have chosen JavaScript as the language I teach. It's therefore very handy to be able to run JavaScript on the Pico. It works great with kaluma, and we use it in combination with this page I made: https://www.mathsuniverse.com/pico (video here: https://m.youtube.com/watch?v=J1tOiVSxBqY&feature=youtu.be )

After this academic year I intend to make all my teaching resources available to other teachers/schools who may want to use JavaScript as well, so it would be a shame if kaluma was no longer an option.

I very much appreciate all the work you've done on it so far, thank you.

babelproject, 2024-10-30:
**We also use Kaluma in Hungary**. I spent over a year with Micropython. Then I started looking for something more usable. I was afraid that I would have to use C because I am not satisfied with the most popular development environments. I accidentally found KalumaJS.

**I rewrote all my Pico W code in javascript**. My Pico W microcontrollers are connected to a Raspberry Pi 4B wifi access point and a server (with KalumaJS Net/Socket API). I switch with Pico W lamps, consumers (max 10 Amper) and large consumers (max 30 Amper). **SWITCHING IS IMMEDIATE!** With Micropython, switching was randomly delayed.

**I also use sensors.** I want to upload my MPU9250 KalumaJS code soon (my code return not just raw datas). I signed up (just now) for Github only because of KalumaJS. I think the advertised popularity of Micropython and Arduino is misleading. The javascript code is straightforward, readable and so beautiful. **For example, I2C is great in KalumaJS**. It was incredible to me when I found that **most of Javascript's built-in capabilities also work great (JSON, Array, Math, Promises, TypedArray and others)**.

**It takes no more than 5 minutes to start programming the microcontroller using KalumaJS**. Most of the time, installing Thonny is acceptable. There is always something wrong with the Arduino installation. VSCode requires a "VSCode university" degree. Everyone knows, but no one complains, because most people do not know KalumaJS yet.

I don't think C is suitable for learning programming in the first place. I didn't find Micropython suitable for Pico. **LEARNING Pico requires Javascript**. Well-known: KalumaJS needs manufacturer support to be popular (included in the Raspberry documentation) or at least it would require a lot of videos on YouTube.

This project is needed by many people (not just for teachers and students). This project is not supported by anyone's money, but it is brilliant.

communix, 2024-10-31:
@jakeg @babelproject 
Thank you for your support of KalumaJS. I was unaware that our project is actively employed in educational and personal development endeavors. I intend to initiate the porting of KalumaJS to the RP2350, the pico 2 board, and release the initial version as soon as possible. As I am unfamiliar with the RP2350, I cannot definitively estimate the required time, but I will certainly endeavor to complete the task.

niklauslee, 2024-10-31:
@communix Thank you for your effort.

babelproject, 2024-10-31:
COMMUNIX: Thank you very much for this decision! I will upload codes for beginners from the beginning of next year.

It would also be useful to make video materials later. 
JAKEG: Some teachers could give their students an exam assignment to create a clever code and post it on YouTube.

closed 2024-11-03 by communix

jakeg, 2024-11-03:
Amazing! Thank you @communix

communix, 2024-11-03:
@jakeg @babelproject I just released [1.2.0-beta.1 version for pico-2](https://github.com/kaluma-project/kaluma/releases/tag/1.2.0-beta.1) board. Please find it.
If you find any issue. please create "issues", I'll check it and fix it ASAP.

jakeg, 2024-11-08:
@communix I've just bought a Pico 2 and tried a very simple inbuilt LED toggle script on it:

```js
let led =  25
pinMode(led, OUTPUT)
setInterval(() => {
  digitalToggle(led)
  console.log('Toggle')
}, 300)
```

... and confirm that it works fine. Thank you!

timrogers, 2024-12-08:
Thanks for the effort here to support the RP2350! Will the beta release work with the Pico 2W as well? 🤔

communix, 2024-12-16:
@timrogers I'll buy Pico 2W and check it. And Kaluma 1.2.0 will support Pico 2 and Pico 2W.

jakeg, 2024-12-24:
I just bought a Pico 2W... I presume [1.2.0-beta.1](https://github.com/kaluma-project/kaluma/releases/tag/1.2.0-beta.1) is the version to try with it?

jakeg, 2024-12-26:
1.2.0-beta.1 doesn't seem to include a `.uf2` which works with the Pico 2W's wireless radios. [kaluma-rp2-pico-w-1.2.0-beta.1.uf2](https://github.com/kaluma-project/kaluma/releases/download/1.2.0-beta.1/kaluma-rp2-pico-w-1.2.0-beta.1.uf2) runs on it, but of course doesn't include wireless drivers. Meanwhile [kaluma-rp2-pico2-1.2.0-beta.1.uf2](https://github.com/kaluma-project/kaluma/releases/download/1.2.0-beta.1/kaluma-rp2-pico2-1.2.0-beta.1.uf2) doesn't run at all.

communix, 2024-12-26:
@jakeg I think kaluma-rp2-pico2-1.2.0-beta.1.uf2 works on pico 2W but it doesn’t have wireless module. I bought pico 2W board but it’ll be delivered mid of January. I’ll implement new code to support poco 2w board once I receive the board.

jakeg, 2024-12-26:
Let me know if you'd like me to post you one for free sooner (I'm in no rush though), from the UK (just pop me your address somehow). I have a Raspberry Pi shop a 10mins walk from my house, so picked one up 2 days ago.

communix, 2024-12-30:
@jakeg Thank you for your kindness but I think it's better off waiting my board.
I built Kaluma for pico2-w board. Could you please test it?
[kaluma-rp2-pico2-w-1.2.0-beta.1_1229.uf2.zip](https://github.com/user-attachments/files/18272581/kaluma-rp2-pico2-w-1.2.0-beta.1_1229.uf2.zip)

jakeg, 2024-12-30:
@communix can confirm that works, thank you :)

communix, 2024-12-30:
@jakeg Did you check wifi functions?

jakeg, 2024-12-31:
@communix yes, it can connect to a WiFi network and serve an http server over it.