kaluma-project/kaluma #657: Can't create HTTP server on wireless access point

jakeg, 2024-12-30:
Can create server when connecting to WiFi, but can't create one when making a wireless access point:

```js
let { PicoCYW43WIFI } = require('pico_cyw43')
let pico_cyw43 = new PicoCYW43WIFI()
let http = require('http')

console.log('Starting...')

pico_cyw43.wifiApMode({ ssid: 'Pico', password: 'password' }, (err) => {
  if (err) console.error('err', err)
  console.log('access point running')

  let server = http.createServer((req, res) => {
    console.log('request', req)
    res.writeHead(200, 'OK', {
      'Content-Type': 'text/plain',
      'Content-Length': 1
    })
    res.write('1')
    res.end()
  })

  console.log('Trying to listen...')
  server.listen(80)
  console.log('Listening on', server._dev.ip)
})
```

Outputs:

```
Starting...
access point running
Trying to listen...
```
... but never manages to listen.

jakeg, 2024-12-30:
This is with both the Pico-W and Pico-2W

communix, 2025-01-06:
@jakeg Thank you for your issue report. I fix this issue. please try it with 1.2.0-beta.2

closed 2025-01-06 by communix