kaluma-project/kaluma #665: WiFi AP unusable on Pico W: WiFi AP not providing any IP via DHCP, even with static IP not working  [CYW43] do_ioctl(0, 262, 13): timeout

tricoos, 2025-01-19:
**Firmware** kaluma-rp2-pico-w-1.2.0-beta.2.uf2
**Hardware** Pico WH

**Error**
[CYW43] do_ioctl(0, 262, 13): timeout

**Symptom**
- Phone connects, does not get IP, disconnects
- Windows PC: connects, does not get IP, stays connected and its MAC is shown on the console every second. However, IP cannot be pinged, because the IP is not associated with any network indicating that Windows as the Phone does not get any IP settings via DHCP. Manually setting IP in Windows also does not help.

**Example code**:

```
const WiFi = require('wifi').WiFi;
const wifi = new WiFi();

wifi.wifiApMode({
    ssid: "MyWifi",
    password: "12345678aB"
}, function (err) {
    if (err) {
        console.error(err);
    }
});

setInterval(() => {
    console.log(wifi.getWifiApClients());
}, 1000);

```
Therefore AP is currently completely unsuable. Overriding the default ip and subnet mask in the code also has no effect at all even with completely different IPs like 10.33.33.1 and netmask 255.255.255.0.

communix, 2025-01-21:
@tricoos I think the error message "[CYW43] do_ioctl(0, 262, 13): timeout" is not related to the this issue. The issue is failure in dhcp server start function.

The error messages is from CYW43 driver code and it's the timeout of internal IO control. I don't know the reason of this message but it seems to be related to your network response time.

The dhcp server start function issue is fixed with my new code. Could you please run your code with the attached FW?

[kaluma-rp2-pico-w-1.2.0-beta.2_wifi_ap_test.uf2.zip](https://github.com/user-attachments/files/18483332/kaluma-rp2-pico-w-1.2.0-beta.2_wifi_ap_test.uf2.zip)
[kaluma-rp2-pico2-w-1.2.0-beta.2_wifi_ap_test.uf2.zip](https://github.com/user-attachments/files/18483333/kaluma-rp2-pico2-w-1.2.0-beta.2_wifi_ap_test.uf2.zip)

tricoos, 2025-01-21:
Still not working.

communix, 2025-01-26:
@tricoos Could you please try it with the attached test FW?
I think the latest pico-lib(2.1.0) has some issues with pico-w board. So I use pico-lib(2.0.0) for pico-w board. Please let me know this works for your board.

[kaluma-rp2-pico-w-1.2.0-beta.2-pico-w-test.uf2.zip](https://github.com/user-attachments/files/18548649/kaluma-rp2-pico-w-1.2.0-beta.2-pico-w-test.uf2.zip)

communix, 2025-02-01:
[kaluma-rp2-pico2-w-1.2.0-beta.2-picosdk.uf2.zip](https://github.com/user-attachments/files/18630628/kaluma-rp2-pico2-w-1.2.0-beta.2-picosdk.uf2.zip)

communix, 2025-02-05:
@tricoos I can't reproduce this issue with the PICO-SDK 2.0.0 now. So I'll close it. I would really appreciate it If you confirm that you can't reproduce this issue with it.

closed 2025-02-05 by communix