kaluma-project/kaluma #666: LED example code not working

Mattijah, 2025-01-20:
I used FW 1.2.0 Beta 2 for Pi Pico W. All successfully set up. I have also managed to flash the example code and verified that it is present by running `.flash -r`. For whatever reason the Pi's led is however not blinking.

Running the below command produces no errors. It runs `load` without issues, but once again the led is not blinking.
```
kaluma flash ./index.js --bundle --shell
```

Code:
```javascript
const led = 25;
pinMode(led, OUTPUT);
setInterval(() => {
  digitalToggle(led);
}, 1000);
```

Any suggestions?

Mattijah, 2025-01-20:
Ah, there is a similar ticket already: https://github.com/kaluma-project/kaluma/issues/624. The example in there works. Perhaps, you guys (or whoever has access to that get started article) could fix this. It is confusing if the code is not even working

communix, 2025-01-21:
@Mattijah Thank you for the issue report. I'll fix the document.

communix, 2025-01-26:
@Mattijah Please check https://kalumajs.org/docs/getting-started and close this issue.

closed 2025-01-26 by Mattijah