kaluma-project/kaluma #668: `StateMachine.put` does not work on v1.2

taktran, 2025-02-14:
Hi, thanks for the great work getting JS on the pico! 🌟 

I have an issue with [v1.2](https://github.com/kaluma-project/kaluma/releases/tag/1.2.0) (as well as all the beta releases) + [neopixel package](https://github.com/niklauslee/neopixel), where it stalls on `np.show();`, which underneath the hood is using `StateMachine.put` (https://github.com/niklauslee/neopixel/blob/main/index.js#L58). However, when I use [v1.1]([v1.2](https://github.com/kaluma-project/kaluma/releases/tag/1.2.0)), `np.show();` does work. I'm also using the pico-w.

Any ideas what changed, that could have caused this?

---

Neopixel issue: https://github.com/niklauslee/neopixel/issues/7

vrz, 2025-02-18:
I have the same problem. Tried on different boards

Blaarkies, 2025-03-09:
Same issue for me, I was moving my program over to a Pico W with the new 1.2 release and it freezes on the `statemachine.put()` method call.

The put() method still works with very small arrays such as `new Uint32Array(4)`, but bigger lists just freeze it without any error output.

We can probably use the SPI hack for neopixels in the meanwhile

communix, 2025-03-09:
I checked this issue and found that TX_FIFO is full. So it's stuck in the pio_sm_put_blocking() function. I'll check the root cause of this issue.

communix, 2025-03-15:
@taktran @vrz @Blaarkies I fix this issue. Could you please try it with the attached image and confirm it's fixed?

[Kaluma_PIO_test_image.zip](https://github.com/user-attachments/files/19267273/Kaluma_PIO_test_image.zip)

Blaarkies, 2025-03-16:
I can confirm, the PIO_test artifact for the pico-w works perfectly!

https://github.com/user-attachments/assets/df792a1f-7b0f-4d3c-bd29-3eeca44a6fa2

I also tested this with a 74 pixels strip, and it seems to work fine so far. Is there a size limit to the buffer in `stateMachine.put(buffer)` at all, or a specific error we should expect to catch?

communix, 2025-03-16:
@Blaarkies I don't think there's a size limitation, PICO PIO has Tx/Rx FIFO and neopixel driver use size 8 (32Bit x 8) Tx FIFO.

vrz, 2025-03-17:
I've tested on the board "Waveshare RP2040-Matrix". It works

https://github.com/user-attachments/assets/330b7d0f-e3a0-4a3b-93ff-72ffaccd6b63

taktran, 2025-03-17:
I've got it working too 🎉 Thanks @communix ! ❤ 

https://youtube.com/shorts/ppkDf6NRFsA

communix, 2025-03-23:
Thank you all. I'll release Kaluma version 1.2.1 with this fix today.

closed 2025-03-23 by communix