kaluma-project/kaluma #675: Are there any plans to port Jerryscript v.3.0.0?

vrz, 2025-03-31:


communix, 2025-04-12:
I checked JerryScript v3.0.0 and start to port it.

vrz, 2025-06-23:
What I've found for now.
Program not starts and terminal hangs after '.flash -w' and then '.load'.
Also, terminal hangs after 'CTRL+D'

[image]

communix, 2025-06-28:
@vrz Thank you. I create #677 and let me check it.

communix, 2025-06-28:
@vrz Could you please check system hang issue with this FW? If you confirm it I'll release beta.2 version

[kaluma-rp2-pico2-1.3.0-beta.1-reset_test_0628.uf2.zip](https://github.com/user-attachments/files/20964068/kaluma-rp2-pico2-1.3.0-beta.1-reset_test_0628.uf2.zip)

[kaluma-rp2-pico2-w-1.3.0-beta.1-reset_test_0628.uf2.zip](https://github.com/user-attachments/files/20964069/kaluma-rp2-pico2-w-1.3.0-beta.1-reset_test_0628.uf2.zip)

communix, 2025-07-04:
https://github.com/kaluma-project/kaluma/releases/tag/1.3.0-beta.2 is released. Please use it and open issues if you find any bug in the beta release.